adamgreig/agg-kicad #36: Create incomplete outputs

ok1hra, 2016-02-13:
Error messages:
Traceback (most recent call last):
  File "/home/dan/kicad/agg-kicad-master/scripts/stickerbom.py", line 454, in <module>
    main(xmlpath, pdfpath)
  File "/home/dan/kicad/agg-kicad-master/scripts/stickerbom.py", line 446, in main
    label_height-14, line.refs)
  File "/home/dan/kicad/agg-kicad-master/scripts/stickerbom.py", line 208, in render
    scale_x = max_w / bound_width
ZeroDivisionError: float division by zero

[bd.pdf](https://github.com/adamgreig/agg-kicad/files/128887/bd.pdf)

adamgreig, 2016-02-13:
Do you have an example file that's causing this problem? My guess is it's finding the required bounding area around components is zero width, causing problems.

ok1hra, 2016-02-13:
if anything is missing, i add

[bd.zip](https://github.com/adamgreig/agg-kicad/files/128890/bd.zip)

adamgreig, 2016-03-12:
Hi @ok1hra, thanks for the bug report! I'm sorry it took a while to get around to fixing it. The problem was that P13 has just the one pad, and when its extent was computed, the pad's size was not taken into account, so it had a zero-sized highlight area, which caused the problem.

I've updated StickerBOM to include the size of the pads when computing the extents, so the problem should now be fixed - and seems to work fine on your PCB.

closed 2016-03-12 by adamgreig

ok1hra, 2016-03-17:
Thanks for fix it, now generate without errors.
But otputs pile up in one page [bd.pdf](https://github.com/adamgreig/agg-kicad/files/177464/bd.pdf).

adamgreig, 2016-03-17:
Please try with `python3` instead of `python`, sounds like you are experiencing issue #45

ok1hra, 2016-03-18:
Confirming you're right, now works correctly, thanks again.

commit 35a667ec (2016-03-12): Update StickerBOM to include pad extents in bounds, fixes #36