adamgreig/agg-kicad #44: sexp not behaving well

cchamilt, 2016-03-10:
```
Traceback (most recent call last):
  File "agg-kicad/scripts/stickerbom.py", line 454, in <module>
    main(xmlpath, pdfpath)
  File "agg-kicad/scripts/stickerbom.py", line 429, in main
    pcb = PCB(sexp.parse(f.read()))
  File "agg-kicad/scripts/stickerbom.py", line 185, in __init__
    self._parse(board)
  File "agg-kicad/scripts/stickerbom.py", line 296, in _parse
    self.modules.append(Module(module))
  File "agg-kicad/scripts/stickerbom.py", line 47, in __init__
    self._parse(mod)
  File "agg-kicad/scripts/stickerbom.py", line 126, in _parse
    self._parse_pad(pad)
  File "agg-kicad/scripts/stickerbom.py", line 151, in _parse_pad
    layers = sexp.find(pad, "layers")[1]
IndexError: list index out of range
```
When I wrap  sexp.find(pad, "layers") usage with a len check, I get a blank 11 page pdf.

adamgreig, 2016-03-10:
Do you have an example .kicad_pcb file that triggers this behaviour?

cchamilt, 2016-03-11:
For this project it would require an NDA as the design isn't mine to share.  Would that be OK?

adamgreig, 2016-03-12:
That is OK if necessary, though if you can isolate what's causing the problem perhaps that would be easier. Could you try adding `print(pad)` to the top of `_parse_pad` and see what the output is? It looks like probably one of the pads in your board does not have _any_ layers which is a bit unusual so might be a parsing issue.

cchamilt, 2016-03-12:
Ah, OK.  I would guess there are some old patterns with through-holes that were set to no layers instead of npth.  I'll add the print and review them.  Hopefully that is the fix.

cchamilt, 2016-03-13:
I went through and located the bad npth.  So everything has layers and the script finishes, but still seems that the pdf is blank.  I'll get you a copy of the NDA asap.

adamgreig, 2016-03-13:
Okay, thanks. Crashing with no layers is obviously still a bug but good to
know that was the cause of the exception.

Are you using Python 2 and if so could you try with Python 3? Should work
with both but is the root of another pdf rendering bug I'm chasing down.

cchamilt, 2016-03-19:
Yeah, I can confirm it doesn't render in either version of python.