adamgreig/agg-kicad #45: StickerBOM disagrees with Teapot

cillian64, 2016-03-11:
StickerBOM does terrible things when exposed to [my "teapot" board](https://github.com/cillian64/teapot/tree/master/board) - it seems like p1 of the output gets moved to and overlayed onto p2.

adamgreig, 2016-03-12:
Works fine in Python 3, issue seems to be in Python 2. Can't spot any obvious issue...

ch-aurich, 2016-03-26:
the problem is the way you generate the label positions... I added some debug outputs to see what happens in my script (until now I added the stickerbom feature only to the development branch: https://github.com/I2SE/eagle-bom.py/tree/develop):

python2 first generates all label positions in function sheet_positions (is initiated by function "zip") and then adds content to them... python3 initiates the generation of the label positions when they are used.

I will try to fix this issue in my script and then will suggest the solution to you as a pull request.

adamgreig, 2016-03-26:
ah, of course, thanks! I think just swapping to the generator zip in `itertools.izip` should solve it.

ch-aurich, 2016-03-27:
just using itertools.izip did not work for me with python 3.x therefore I implemented it a bit different... you can have a look at http://codereview.stackexchange.com/questions/26271/cleaner-way-to-import-izip-for-different-versions-of-python where I found this solution to be the most elegant...

adamgreig, 2016-03-27:
Fixed by #46

closed 2016-03-27 by adamgreig