adamgreig/agg-kicad #58: 3D Models

HarkonenBade, 2016-06-18:
Would it be possible to either add 3D models for parts, or at least for standard things like 0603/1206 style packages to reference the standard kicad library models for those parts?

adamgreig, 2016-06-18:
It would be really nice! I'm holding out on KiCAD supporting more 3d formats first, as the current format is a quite weird Wings3D thing whereas work is underway to support STL and other formats, at which point most parts can be downloaded from the manufacturer and the rest can be defined in text easily.

It might be relatively easy to refer to existing KiCAD 3d parts (eg an extra optional field in `build_mod_chip.py` and `build_mod_ic.py`) but I've not looked in to it yet...

HarkonenBade, 2016-07-11:
I've started collating some of my work on bits like this in https://github.com/HarkonenBade/kicad-models

Unfortunately I think until kicad supports more reasonable model formats, or someone writes a VRML exporter targeted at kicad, I don't think programmatic generation of models is going to be reasonable. Also I've been using openscad for a lot of stuff and really it needs better support for specifying colors/materials of objects.

So for now I would set this issue to awaiting upstream changes.