adamgreig/agg-kicad #67: Add an arg to which field to print instead of "supplier":"number"

metropt, 2016-07-01:
Hi,
Thank you for the work doing the script :)

I don't usually have the supplier on the part. Only the manf. part number which I call "manf#" because I use [KiCost](https://kicost.readthedocs.io/en/latest/readme.html) to generate the BOM with prices, etc...

So to use your code I needed to comment the condition on line 473 which works nice :) I would be able to create the PDF however the supplier field doesn't print the same value on all components but some random field.

It would be nice to be able to select the field you want to use.

I'll try to do that.

HarkonenBade, 2016-07-01:
You might want to take a look at #66

metropt, 2016-07-01:
Thank you, edited supplier_to_output and add only manf# solved my problem!

adamgreig, 2016-07-02:
Great, glad to hear you've worked it out. Once #66 lands you can do it from the command line in KiCAD without editing the script at all which will be nice.

closed 2016-07-02 by adamgreig