adamgreig/agg-kicad #123: stickerbom doesn't work anymore with nightly builds

mminnella, 2017-08-26:
Seems that stickerbom it's unable to work with nightly build releases, outputted pdfs are blank, maybe it's due to some recents changes on kicad_pcb formats?

Best Regards

adamgreig, 2017-08-26:
I've just tried with kicad `201708261203+db3491f~59~ubuntu16.04.1` and it seems to work OK. Could you share a kicad_pcb file that isn't working for you? Also, what version of Python and PyCairo are you using, on what OS?

mminnella, 2017-08-28:
Dear Adam,
Here's you could find a copy of my kicad_pcb and xml files:

https://www.dropbox.com/s/pimnax0q1w0epbp/VEAMv2_outline.tar.gz?dl=0

These are my system details:

Ubuntu 16.04.3 LTS
kicad 201708271203+e3c64f1~59~ubuntu16.04.1
python-cairo is already the newest version (1.8.8-2)
python3-cairo is already the newest version (1.10.0+dfsg-5build1).

i've just tried with both python / python3 , stickerbom output a one page
blank pdf file.

Best Regards

Massimiliano

adamgreig, 2017-08-28:
How are you calling `stickerbom.py`? I think you will need to pass in `--suppliers "manf#"` since by default it only emits stickers for line items with fields for the default suppliers: `Farnell`, `RS`, `DigiKey` or `Digikey`, and `Mouser`. But if you pass `--suppliers` you can tell it what to use instead, like `manf#` or `farnell#` for your file.

I just tried and it seems to work OK.

mminnella, 2017-08-28:
You're right! , sorry, it  was my fault !

Best Regards

Massimiliano

closed 2017-08-28 by adamgreig