adamgreig/agg-kicad #143: 3D models have all moved

russss, 2019-11-30:
Unless I'm completely mistaken, the 3D models we reference have moved locations in new(ish) versions of KiCAD... e.g.:

`Housings_SOIC.3dshapes` is now `Package_SO.3dshapes`
`Resistors_SMD.3dshapes` is now `Resistor_SMD.3dshapes`

But worse, the name of the actual files has also changed in many cases. This seems to have happened back in Feb 2018.

adamgreig, 2019-11-30:
Yep, it's pain... Most of them will have to be updated by hand I think. I'm also not convinced defaulting to resistors for all chip components is the best bet, but I don't really want to split each size into resistor vs capacitor footprints. On the other hand, maybe it would allow different silk for each and perhaps slightly different pad geometries...

Also, for a lot of parts there are no models in KiCAD itself, but there are manufacturer supplied models, which might be worth including in the repo.

adamgreig, 2019-11-30:
On the whole I think the project needs a bit of TLC; I still use it exclusively for all my PCB designs but my impression is that the stock KiCAD libraries are much higher quality today than when I started agg-kicad, so I think other people probably have less need for it. That said, I think the scripted symbol and footprint generation is still really useful. I wonder if it's worth rejigging to make it easier to use the generator scripts or generated parts individually (perhaps part of #121 or something), or something else.

HarkonenBade, 2019-12-01:
I'm still a big fan of the library in general and have been using it myself for a recent major project. I agree that the stock libraries seem to be improving dramatically, but I do find the scripted yaml generation to be the best way to throw together new parts quickly and accurately.
I'd be very in favour of seeing a setup where a given project has a single yaml (or maybe a small local directory of them) that can be run through a single script to generate a symbol library and footprint library of parts for that project. Particularly it might be interesting at looking at integrating the library and footprint generation. Given it's not entirely unusual to be generating new QFN/DFN variants for specific ICs, and it may be nice to be able to actively just place the footprint definition inside the library symbol definition for that specific part.

Also back on the models, I have some current updates that i can upstream to restore some of the models, and I had a little work on scripted model generation that was abandoned at the time due to kicad's limited model types support, but may now be worth renewing.

Gasman2014, 2019-12-01:
Would https://github.com/qeda/qeda be a suitable scriptable yaml generator?