adamgreig/agg-kicad #148: Does stickerbom work in KiCad 6.0

Phozer, 2022-10-18:
First of all, great Project!

I tried using the stickerbom.py to create labels to stick on my component bags. I face some problems using it, so I want to ask if it even works with KiCad 6.0? 

This is what i get as output. 
```
python C:\03_Data\repos_external\git\agg-kicad\scripts\stickerbom.py C:\03_Data\repos\SVN\TestAutomation\TestAdapter\SNIFX\SNIFX.QP2\2_KiCad\SNIFX.xml C:\03_Data\repos\SVN\TestAutomation\TestAdapter\SNIFX\SNIFX.QP2\2_KiCad
Traceback (most recent call last):
  File "C:\03_Data\repos_external\git\agg-kicad\scripts\stickerbom.py", line 589, in <module>
    main()
  File "C:\03_Data\repos_external\git\agg-kicad\scripts\stickerbom.py", line 534, in main
    pcb = PCB(sexp.parse(f.read()))
  File "C:\03_Data\repos_external\git\agg-kicad\scripts\stickerbom.py", line 166, in __init__
    self._parse(board)
  File "C:\03_Data\repos_external\git\agg-kicad\scripts\stickerbom.py", line 306, in _parse
    min(m.at[0] for m in self.modules),
ValueError: min() arg is an empty sequence
```

adamgreig, 2022-10-18:
Thanks for opening the issue! It looks like KiCad 6 made a few changes that broke stickerbom, mostly renaming "modules" to "footprints" in the board file and changing how graphic arcs are represented from "centre, start, angle" to "centre, mid, end". I've just pushed an update to stickerbom that seems to work for me on some KiCad 6 boards, could you check if it works for you?

Phozer, 2022-10-19:
Thanks for your quick fix. 
Seems to work much better than before. 
Now i have this issue: 
```
C:\03_Data\repos_external\git\agg-kicad\scripts>python stickerbom.py C:\03_Data\repos\SVN\TestAutomation\TestAdapter\SNIFX\SNIFX.QP2\2_KiCad\SNIFX.xml C:\03_Data\repos\SVN\TestAutomation\TestAdapter\SNIFX\SNIFX.QP2\2_KiCad\stickerBOM.pdf
Traceback (most recent call last):
  File "C:\03_Data\repos_external\git\agg-kicad\scripts\stickerbom.py", line 600, in <module>
    main()
  File "C:\03_Data\repos_external\git\agg-kicad\scripts\stickerbom.py", line 596, in main
    cr.show_page()
__main__.cairo.MemoryError: out of memory
```
I tried with different boards. A PDF gets created but it's blank.

adamgreig, 2022-10-19:
Hm, that's odd. It seems like this might be an issue with PyCairo on Windows, see https://github.com/pygobject/pycairo/issues/239. Perhaps you could try the builds linked there, or a Linux/WSL build?

Phozer, 2022-12-05:
Do you have e public project which worked for you? Then I can check for differences and troubleshoot. Would love to get this to work.

Phozer, 2022-12-06:
Is there an option to uses the copper layer instead of the fab layer for the drawings?

adamgreig, 2022-12-08:
Ah cool, glad you figured it out! There's not a built-in option at the moment, but if you change `F.Fab` for `F.Cu` and `B.Fab` for `B.Cu` on lines 579-593 that should do it; currently it uses F.Fab first but if a footprint doesn't have any items on the fab layer it will instead render use the copper and silkscreen layers.

Phozer, 2022-12-16:
I will now close this Issue because for me it's fixed now. I just noticed that my Board outline did not get drawn. Just if you bother you may fix that. Thanks for your work. It helped a lot.

closed 2022-12-16 by Phozer

adamgreig, 2022-12-19:
Thanks for sticking with it, I'm glad it's working for you now! Not sure what's up with the board outline, I'll try and look into it.